Improbable-AI/walk-these-ways #1: Stopped after `sudo make autostart`

Kashu7100, 2022-12-02:
Thank you for sharing the code for this awesome work. 

I followed the setup in the README to set up your code and I would like to have some advice on the `Running the Controller` section. After running the second command (`cd ~/go1_gym/go1_gym_deploy/docker; sudo make autostart`), the code seems to stop after a while (please refer to the attached image).

I pressed [R2] several times at this moment but it did not do anything.

[image]

Do you have any suggestions about this situation?

Sincerely,

Kashu7100, 2022-12-03:
After `python3 deploy_policy.py` command, the process seems to stop at `pkl_cfg = pkl.load(file)` 

[image]

Kashu7100, 2022-12-03:
After reading your paper, I noticed that my Go1's jetson board is normal nano (not NX).
I think maybe the lack of computation power of my Go1 is causing this issue. 

Do you think running without docker can help? 
Also, I want to check whether I can run the policy from my PC (not the onboard nano). I am wondering if there is a way to communicate with `lcm_position` from outside of Go1 via ethernet cable.

gmargo11, 2022-12-03:
Odd! But the `parameters.pkl` file is present at `/home/isaac/go1_gym/runs/gait-conditioned-agility/pretrain-v0/train/025417.456545/parameters.pkl`, right? 

You could try ensuring this file is fully readable by all users by navigating to `runs/gait-conditioned-agility/pretrain-v0/train/025417.456545/` and running `chmod +777 parameters.pkl`

Re: Other questions
- The controller should run fine on a normal Nano, NX is not necessary. Also, this line is just loading a small file using the `pickle` library which shouldn't be computation-intensive.
- You can run the whole controller outside docker, but you would need to install the dependencies on your Nano manually. Docker is easier for sharing code across systems. A simpler debugging step might be to first ensure that this `pkl.load` command can run successfully outside docker.
- I haven't tried networking an external PC with the Go1. But it's definitely possible. The deployment script communicates with `lcm_position` by sending and receiving LCM messages. So you just need to ensure that the LCM messages published by your PC are received on the Go1. You might find some information here: https://lcm-proj.github.io/

Kashu7100, 2022-12-05:
Thank you for your advice. Actually, `chmod +777 parameters.pkl` works. 
After the `chmod` and starting the controller manually, the controller successfully works. 

It seems like there are some stability issues with running the code, but I will look deeper!
Also, it would be helpful if you can publish the source code for `lcm_position` file.

Thank you a lot.

gmargo11, 2022-12-05:
Great to hear that resolved the issue. I've just made the file permissions correct in the repo. I also added the source code for the `lcm_position` file.

Closing this issue since I think we've eliminated the original error. If you open a new issue with any stability problems you're referencing, happy to take a look. 

Also, would love to see video if you take the system out for a successful walk :-) Thanks for being an early tester of the deployment code

closed 2022-12-05 by gmargo11

Kashu7100, 2022-12-06:
Thank you for adding the source code for `lcm_position`.

It seems the following header file is required to compile:
```
#include "state_estimator_lcmt.hpp"
#include "leg_control_data_lcmt.hpp"
#include "pd_tau_targets_lcmt.hpp"
#include "rc_command_lcmt.hpp"
```
Some of them seemed like from the Cheetah Software but I could not figure it out.

Also, I will definitely post the video!