Improbable-AI/walk-these-ways #3: ImportError: cannot import name 'PrefixProto' from 'params_proto'

manipulability, 2022-12-09:
Thank you for sharing your awesome code!

After installing the isaac gym and go1_gym, the `ml-dash` worked fine. However, when I ran `python scripts/test.py`, I have error messages like this:
```
File "/home/walk-these-ways/go1_gym/envs/base/legged_robot_config.py", line 3, in <module>
    from params_proto import PrefixProto, ParamsProto
ImportError: cannot import name 'PrefixProto' from 'params_proto' (/home/venv/walk-these-ways/lib/python3.8/site-packages/params_proto/__init__.py)
```

After that, I updated `params_proto` via `pip install params-proto==2.10.0`, and then I confirmed that `python scripts/test.py` works well,, but I have other error messages as below when I ran `python -m ml_dash.app'
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/venv/walk-these-ways/lib/python3.8/site-packages/ml_dash/app.py", line 5, in <module>
    from params_proto import cli_parse, Proto
ImportError: cannot import name 'cli_parse' from 'params_proto' (/home/venv/walk-these-ways/lib/python3.8/site-packages/params_proto/__init__.py)

```

Have you ever experienced such these issues?

I ran your code on Ubuntu 22.04, python3.8.

manipulability, 2022-12-09:
Oh, I jush solve my issue.

I confirmed `ml-dash' works well, after updating `pip install ml-dash==0.3.25'

closed 2022-12-09 by manipulability

developermastermind07, 2023-05-02:
@manipulability I am also facing same problem. Can you help me out. 
Following are the details:
OS: Ubunut 18.04
params_proto 2.10.5
ml-dash-app 0.3.20