Improbable-AI/walk-these-ways #7: Minor Fixes

TextZip, 2023-01-30:
@gmargo11 thanks for the amazing repo, it was very intuitive to follow. Looking forward to updates on `Deploying a Custom Model`.  [Here](https://youtu.be/4cc_2OeSXDw) is a video of Go1 running around with the policy. 
  
Here are some minor fixes that could make deployment easier:

### Installing the Deployment Utility

1. The `cd go1_gym_deploy/scripts && ./send_to_unitree.sh` downloads the `deployment_image.tar` to the scripts folder. While in the next step `cd ~/go1_gym/go1_gym_deploy/installer && ./install_deployment_code.sh` the script searches for the `deployment_image.tar` in  `/go1_gym/go1_gym_deploy/installer`.

### Running the Controller

1. There seems to be no safe way to exit the policy, the policy continued to run even after the script was terminated. 

### Some information that might be helpful for others

1. If you run out of disk space during `cd ~/go1_gym/go1_gym_deploy/installer && ./install_deployment_code.sh` consider changing the script to use 192.168.123.13 instead (at least in my Go1 Edu with 3 Jetson nano, I only had the required disk space to copy the tar and extract the image in only 192.168.123.13).  Alternatively, consider deploying on an external PC. 

2.  When deploying on an external PC/NUC, if you get the following error 

```walk-these-ways/go1_gym_deploy/autostart on  master [!?] 
➜ sudo ./start_unitree_sdk.sh
./lcm_position: 1: Syntax error: word unexpected (expecting ")")
```

It is likely because the ./lcm_position has been compiled for  `ARM aarch64` (to run on the jetson), please recompile it for your architecture(external PC/ NUC) using https://github.com/Improbable-AI/unitree_legged_sdk.

gmargo11, 2023-02-01:
@TextZip Thrilled to see the policy running on your Go1! Thanks for sharing the video.

Will you consider submitting a pull request with these fixes and updates to the README? I would be happy to merge it so that you are attributed :)

-Gabe

gmargo11, 2023-03-11:
Updated the readme and `install_deployment_code.sh`!

gmargo11, 2023-03-11:
Updated the readme and `install_deployment_code.sh`!

48267a39f1ef05fce2f6bbe18a95414f90bc6264
5b18170e2928b34452c3320254362897fd3d82d7

closed 2023-03-11 by gmargo11

TextZip, 2023-07-09:
Hi @MariaBana , 

I never faced this issue so I'm not entirely sure but did you check this out? 
> This is probably due to the fact that sudo privileges were used during the installation.
I was installing nvidia drivers and nvidia container toolkit with sudo and when I tried to run the docker container without sudo, I got the exact same error.

Source: https://forums.developer.nvidia.com/t/nvidia-container-cli-initialization-error-load-library-failed-libnvidia-ml-so-1/237759/2