Improbable-AI/walk-these-ways #12: About actuator net data collection rate

yangyichu, 2023-02-27:
In the training script it seems the joint data were collected at 50Hz, while compute_torque() in isaac_sim actually calculates torque at 200Hz (dt=0.005, though the target stays same for 4 iteration but joint state error is changing). Will this collect frequency influence the performance of actuator net?

gmargo11, 2023-02-27:
Hi @yangyichu ,

Yes, thanks for the careful inspection; your interpretation is correct! The actuator network was trained at 50Hz, but during forward sim it's being evaluated at 200Hz. Since the network takes a short history of error values, this might impact performance, I'm not sure how much. I'm going to implement a fix, if the policy trains and deploys well I'll update the code.

closed 2023-04-26 by gmargo11

ammousa, 2025-07-16:
Hi @gmargo11,
Any update on this point? it is very confusing and I'm wondering how the physics are consistent which seems unlogic to me