Improbable-AI/walk-these-ways #13: Different Go1 URDF with original unitree repository

anahrendra, 2023-03-01:
Hi! I just noticed that the URDF that you used in this repo is different with the original URDF (https://github.com/unitreerobotics/unitree_ros/blob/master/robots/go1_description/urdf/go1.urdf). For example, the max. effort in the original repo is 23.7, while yours is 33.5. Is there any reason for that?

Thanks in advance! Have a nice day

gmargo11, 2023-03-11:
Hi @anahrendra ,

Hmm, I obtained this URDF from Unitree before they uploaded the linked file. Seems like several updates have been made to the official model as recently as November. I'll try out their most recent official model and update this repo if it works fine.

Appreciate you noticing this and bringing it to my attention!

Gabe

Calvindrinks, 2023-03-20:
Torque 23.7 upper limit is also working in my project. 33.5 is now the setting for a1.
The new version of urdf added the description of the sensors. Considering that it may affect inertia, I did not remove them.

By the way, if you are going to modify from the new version of urdf, make sure to set `<dynamic damping="0" friction="0"/>`. This seems important, otherwise you will have bad results after training.

closed 2023-04-26 by gmargo11

Haichao-Zhang, 2023-06-28:
> Torque 23.7 upper limit is also working in my project. 33.5 is now the setting for a1. The new version of urdf added the description of the sensors. Considering that it may affect inertia, I did not [...]
 
Can you explain a bit more on why setting ``<dynamic damping="0" friction="0"/>`` is important for obtaining good training results?
Thanks!

gmargo11, 2023-06-29:
@Haichao-Zhang as far as I'm aware, it's a bug or limitation in how IsaacGym reads the URDF.

Haichao-Zhang, 2023-06-29:
@gmargo11  OK got it. Thanks for sharing!

GuoPingPan, 2023-07-09:
I have a question about the stiffness and damping of isaacgym. Why the api `get_asset_dof_properties` shows that all the joints' stiffness and damping is 0? Is it no need to set these two params? If it will cause sim2real gap?