Improbable-AI/walk-these-ways #15: Basic Deployment Question

dhruvmetha, 2023-03-20:
Hey, I had a basic question regarding the deployment. In the low level mode of the robot, is the robot set to a walking joint configuration when hung in the air, and then placed on the ground for the remote commands to work?

Not sure if this is the right place for this query, if not, I'm sorry and I will drop an email with this question.

dhruvmetha, 2023-03-21:
Okay, I see that the controller calibration (R2) moves the robot from nominal position to standing position and then we start the control loop (R2).

gmargo11, 2023-03-31:
Hi @dhruvmetha ,

Before running low-level control mode, I usually follow these steps (from https://github.com/Improbable-AI/walk-these-ways#running-the-controller--):

> Place the robot into damping mode. The control sequence is: [L1+B], [L1+A], [L1+L2+START]. After this, the robot should sit on the ground and the joints should move freely.

After the robot is in this damping mode and sitting on the ground, then I start the low-level controller. The robot could also be hanging from the support beam when the controller starts.

closed 2023-03-31 by gmargo11