Improbable-AI/walk-these-ways #19: deploy in amd64

BoyuanDeng, 2023-04-08:
Hi! bro, i want do try deploy this work in my NUC but nano in go1, so could you please compile lcm_position.cpp in amd64 or send to me state_estimator_lcmt.hpp....(I don't want this code, just  don't want to trouble you),  you can send compiled lcm_position to dengboyuanalan@gmail.com
thank you!

closed 2023-04-08 by BoyuanDeng

gmargo11, 2023-04-08:
Hi @BoyuanDeng ,

You can compile https://github.com/Improbable-AI/walk-these-ways/blob/master/go1_gym_deploy/lcm_types/state_estimator_lcmt.lcm into `state_estimator_lcmt.hpp` using `lcm-gen -x state_estimator_lcmt.lcm`, see here for more details: https://github.com/lcm-proj/lcm/blob/master/docs/content/tutorial-cpp.md

-Gabe

BoyuanDeng, 2023-04-09:
Hi Gabe, so happy that get your reply,thanks!( i success compiled in yesterday)

邓博远
***@***.***