Improbable-AI/walk-these-ways #21: Power protection issue

dplozza, 2023-04-26:
Hi, I'm trying to test this repository on a Go1 robot and I encountered a problem seemingly related to the power protection.

In the README you warn about the protection system cutting off power during fast running because of PowerProtect level 9. However, in my case, this happens very frequently even when the robot is moving relatively slowly.
Interestingly, when that happens I don't get an "Error: Power Protection" message (which instead happens when I set a lower PowerProtect level, as expected), also I lose ssh access and all the boards seem to reboot. This seems to indicate that a hardware power protection/failure is happening instead of the software power protection.
This was happening every time I ran the default `lcm_position` program, I had to reduce the initial P gain in order to be able to even start the controller.

I contacted Unitree, but they have not been really helpful, so I wanted to ask if you also encountered this problem.

gmargo11, 2023-05-02:
Hi @dplozza ,

I haven't encountered this problem, but here are some questions to help debug:

1. Are you placing the robot into damping mode before running `lcm_position`? (https://github.com/Improbable-AI/walk-these-ways#running-the-controller--) The robot should lie down on the ground and its joints should move freely after this.
2. Does the default `example_position` script from https://github.com/unitreerobotics/unitree_legged_sdk work? 
3. Is the robot overheating? Check by touching the motors and the metal plate by the USB & ethernet ports. If these are very hot, make sure the vents on the bottom of the robot are not blocked, that the fans are running (emitting some air from the vents), and the room is not too warm.
4. Try running `lcm_position` from another board (`192.168.123.14` instead of `.15`)

By the way, to clarify, are you saying that the program does not fail if you decrease the P gain?

-Gabe

dplozza, 2023-05-03:
Thank you @gmargo11 for the response.

To clarify, if I decrease the initial P gain in `lcm_position` the problem only appears when the robot joints are moving too fast.

After further testing, the problem mainly appears while using an external power supply, and it is caused by too high transient currents when many motors a (without triggering the software power protection). 
This was unexpected since it is a 700W power supply that I have been using with the A1 without any problems. So it seems that the Go1 causes higher transient currents,

When using the battery there still are some failures, but in that case, they seem to be actually caused by the software power protection, as you warn in the documentation.

closed 2023-05-03 by dplozza