Improbable-AI/walk-these-ways #29: Build docker container for different version of JetPack

Vassil17, 2023-06-08:
Hi,

I would like to build the docker container on my own for a slightly different JetPack version (our Go1 has L4T R32.4.4). Did you install ROS (melodic) on it (as those commands are commented out and there's no install_scripts/install_ros.sh in the repo)?

I am getting some incompatibility issues when I try mixing pytorch with ros melodic, so I was wondering if/how you managed to get it to work? 

Thank you!

gmargo11, 2023-06-09:
Hi @Vassil17 ,

It shouldn't be necessary to install ROS in the deployment docker container for this project. You should be able to build the Dockerfile as-is on your Jetson, modifying line 3 (https://github.com/Improbable-AI/walk-these-ways/blob/master/go1_gym_deploy/docker/Dockerfile#L3) to match your L4T version.

I haven't tried mixing ROS with this version of the codebase, so unfortunately am not familiar with the dependency issues that might arise if you try adding it. If you need ROS, ideally, maybe you can find a pre-built base image for Jetson that has ROS and l4t-pytorch both installed? But I haven't looked

-Gabe

closed 2023-06-09 by gmargo11

Vassil17, 2023-06-13:
> Hi @Vassil17 ,
>
> It shouldn't be necessary to install ROS in the deployment docker container for this project. You should be able to build the Dockerfile as-is on your Jetson, modifying line 3 [...]

Hi @gmargo11,

Thanks, that seems to work!

For others stumbling upon this, I couldn't get ROS melodic to work with pytorch, possibly due to the python 2 vs python 3 requirements that the two have.