Improbable-AI/walk-these-ways #32: How to run a newly trained network on the Go1?

Robokan, 2023-06-19:
I have trained a new network and want to deploy it onto my Go1. It looks to be using rsync to move all the runs to the go1. How do I get it to choose my run rather than pretrain-v0? (which I am assuming it automatically uses).

Robokan, 2023-06-19:
Found it. Line 73 in go1_gym_deploy/scripts/deploy_policy.py

closed 2023-06-19 by Robokan

Qianzhong-Chen, 2023-06-21:
Hi, could you please be so kind to tell me how to make the self-trained policy (a .pt file) be available to be deployed with code of this repo? If my guess is correct, deploy_policy.py mainly uses the two .jit files (adaptation_module_latest.jit and body_latest.jit).

Robokan, 2023-06-23:
I haven't done it yet but as far as I can tell all you do is 
1) train a new network. Then the result will be in the "runs" directory. Then 

2) edit: deploy_policy.py to use your run rather than the pretrained one. 

3) follow the instructions to deploy to the go1. One of the scrips contains rsync I believe it will copy all the runs over to the go1 (including yours)