Improbable-AI/walk-these-ways #34: How to get joint torque values from StateEstimator? Maybe tau_est?

Robokan, 2023-06-19:
In StateEstimator you get back 3 sets of leg values: joint_pos, joint_vel, and tau_est. Is tau_est the force feedback from the joints? If not do you know if its possible to get this information from the Go1?

I am getting ready to put my own neural network that takes all it's movements from motion capture. Your repo has been super helpful. Thanks for posting it.

gmargo11, 2023-06-22:
Hi @Robokan ,

Yes, `tau_est` is the torque estimate for the joints. This estimate comes from Unitree's SDK. It is computed based on the motor current, which can be noisy but seems reasonable in my experience.

Glad to hear you've found the repo useful!

-Gabe

closed 2023-06-22 by gmargo11

Robokan, 2023-06-23:
I see you are also using an additional NN to predict the internal dynamics of the actuators. Did you train this yourself? I would love to see the code for that. I am interested in changing it to output positional values rather than torques. It seems like it would be more efficient. Have you tried it?

gmargo11, 2023-06-29:
@Robokan I haven't tried that! However, I think it's better to provide the torques to IsaacGym so it can compute the forward dynamics. If you set the position directly with a neural network, IsaacGym can't ensure that the laws of physics are maintained.

-Gabe

Robokan, 2023-06-29:
Keeping the computation of the spring/damper in the internal physics engine should be more stable and work with higher integrations steps. So wondering if tuning the stiffness and damping parameters first then using the actuation network just capture and dynamics of the physical actuators might be and interesting approach. It would take in the desired targets and emit desired targets (or just additional forces) to make the isaac gym spring/damper system behave more like reality.

gmargo11, 2023-07-03:
@Robokan that's a neat idea, I haven't tried it! If you have the neural network predict the position target for PD control, I agree it might improve the stability of the physics simulation.

As another extension, you could even train the actuator network to predict the "effective" PD gains that predict the torque profile between time steps. To capture those fast dynamics, you could probably increase the logging frequency to a higher rate than 50Hz. That would give you some data to estimate the effective PD gains

Robokan, 2023-07-04:
Thanks, I will take a look at that and maybe even get the actual simulator in the training loop and see if it can be tuned to be as close as possible.