Improbable-AI/walk-these-ways #35: How to make it automatically run the repo's NN rather than the original on boot?

Robokan, 2023-06-26:
I've trained my own NN and want to have it automatically run when the Go1 is turned on. How would you go about doing this?

Ideally it would be great to be able to toggle between both control systems (mine, and original).

gmargo11, 2023-06-29:
Hi @Robokan ,

To autoboot the learned controller, I usually follow these steps:

1. Disable the built-in controller:

Enter the Raspberry Pi (`ssh pi@192.168.123.161`). 
Navigate to the `sportMode` folder (`cd ~/Unitree/autostart/sportMode`). 
Open the `sportMode.sh` file and comment out every line.
Now, after rebooting the robot will no longer stand up automatically and enter the default "sport mode". You can still activate it if you want by pressing `L2+A`.

2. Make the custom NN autostart using `crontab`.

The `crontab` can be used to execute a script upon boot. You can see some examples here: https://www.geeksforgeeks.org/crontab-in-linux-with-examples/

Enter the body Jetson (`ssh unitree@192.168.123.15`)
Open the `crontab` for editing (`crontab -e`) and add the line `@reboot sh /home/unitree/go1_gym/go1_gym_deploy/autostart/start_unitree_sdk.sh`. This will autostart the low-level PD controller script.
Open the `sudo crontab` for editing (`sudo crontab -e`) and add the line `@reboot sh /home/unitree/go1_gym/go1_gym_deploy/autostart/start_controller.sh`. This will automatically start the docker container and run `deploy_policy.py`.

After step (2), when you reboot the robot, the learned controller will be started after about 1-2 minutes (then you'll need to press `R2` to stand up). NOTE that after step (2), you won't be able to switch into Unitree's built-in controller anymore without removing those entries from the crontab again. This is because sport mode and the unitree sdk shouldn't run at the same time. I don't have a way of toggling between them automatically.

-Gabe

closed 2023-06-29 by gmargo11

Robokan, 2023-06-30:
Thanks! I will try it out. 

Out of curiosity how did you get so much knowledge about the Go1? are there any links to documentation you would recommend?

Robokan, 2023-10-03:
I wasn't able to get this working. The robot just never gets up. Nothing happens. How would I got about debugging this? Are there boot logs? or a way to print the status of each of these operations?

Thanks again for all you help on this.