Improbable-AI/walk-these-ways #41: decimation of 4 seems to create some instability have you been successful with lower values like 2?

Robokan, 2023-07-04:
My networks seem to get a bit shaky (lots of random vibration) at a decimation of 4 (I am using a different NN based trained on motion capture). Have you tried lower ones like 2? Wondering if there are any problems with the go1 keeping up with such a small dt.

gmargo11, 2023-07-06:
Hi @Robokan ,

In my experience, increasing the simulation frequency can improve stability but slow down the simulator. If you want to maintain the default control frequency (50Hz = 0.02s per step), you could, for example, halve the default [simulation timestep](https://github.com/Improbable-AI/walk-these-ways/blob/master/go1_gym/envs/base/legged_robot_config.py#L403) (0.005->0.0025) and double the [decimation](https://github.com/Improbable-AI/walk-these-ways/blob/master/go1_gym/envs/base/legged_robot_config.py#L218) (4->8), which would increase the simulation frequency from 200Hz to 400Hz. I didn't find this necessary when using low PD gains, but at higher gains or with a different type of controller this could become helpful. Btw, are you using PD control or torque control mode?

I found that having a nonzero joint friction or damping can incur extra shaking / instability in isaacgym, which is why I removed them from the URDF. But haven't tested armature, and thanks for the useful note, I may try it sometime.

-Gabe

closed 2023-07-06 by gmargo11

breakds, 2023-07-06:
> My networks seem to get a bit shaky (lots of random vibration) at a decimation of 4 (I am using a different NN based trained on motion capture). Have you tried lower ones like 2? Wondering if there are any problems with the go1 keeping up with such a small dt.
> 
> I also noticed you set armature=0. If I increase it to 0.01 as found in the mujco go1.xml file supplied by unitree the simulation seems to get more stable.

Hi @Robokan,

When you are saying the policy gets a bit shaky - does that happen in the simulator or on the physical Go1 robot deployment? Which simulator are you using?

I am asking because I've seen shaky behavior when deploying some other policies. I do not see shaky behavior when deploying the provided checkpoint.

Robokan, 2023-07-07:
I have my own code base that uses motion capture to train the NN. Have then been putting it into this code base and testing it out. Generally it works in the Isaac simulator part, but fails when transferred to the physical robot. It tends to shake a bit in Isaac sim with decimation of 4 and armature of 0. If I used decimation of 2 it gets more stable and even better with an armature > 0. Still I haven't got it to transfer to the real robot with out shaking a lot.

breakds, 2023-07-07:
> I have my own code base that uses motion capture to train the NN. I have been putting it into this code base and testing it out. Generally it works great in the Isaac simulator part, but so far fails when transferred to the physical robot (It begins shaking uncontrollably). It tends to shake a bit in Isaac sim with decimation of 4 and armature of 0. If I use decimation of 2 it seems to get more stable and is fine if armature > 0
> 
> I am using the PD controller with a p_gain=30 and a d_gain=0.5

Thanks a lot for the information! A smaller `p_gain` can probably tame the shaking on the physical robot I guess. I have been using decimation = 4 and 200 Hz simulation in Mujoco on Go1, with `p_gain` = 50. The learned policy works fine in the simulator but when deployed on the physical robot it shakes a lot. The shaking does not happen on the provided checkpoint though.

Robokan, 2023-07-08:
It looks to be a damping issue. If I lower the decimation to 2 and then increase d_gain to 1.0 it become stable. However at decimation 4 if I increase d_gain it makes it even more unstable.

Robokan, 2023-07-08:
Although something else may be at play too. If I use isaac gym's internal stiffness/damping and use set_dof_position_target_tensor directly setting stiffness: 30.0 and damping to 1.0. I get a different result than when using _compute_torques and set_dof_actuation_force_tensor. Shouldn't they technically be equivalent?

Vassil17, 2023-07-10:
> My networks seem to get a bit shaky (lots of random vibration) at a decimation of 4 (I am using a different NN based trained on motion capture). Have you tried lower ones like 2? Wondering if there [...]

Hi @Robokan ,

Something I noticed that could be interesting to you, if you set armature to a non-zero value it seems to cause some clash with whatever joint friction/damping value you implement. In my case it results in much lower friction in the joints, which I don't think physically should be the case.

breakds, 2023-07-10:
> Although something else may be at play too. If I use isaac gym's internal stiffness/damping and use set_dof_position_target_tensor directly setting stiffness: 30.0 and damping to 1.0. I get a [...]

The position control is done by an actuator which in this case is a PD controller. Does Isaac Gym implement the controller in a different way? For example in Mujoco the damping and Kd are implemented differently but approximately they are describing the same thing.

Robokan, 2023-07-11:
In Isaac Gym I believe it is the same spring/damper system just built into the simulator so it should be much more stable. I would imagine getting a slightly different result but in my case it can't even walk if I change them and I have trained my model with lots of randomization. Its incredibly adaptable to incredible changes just not to changing out the spring/damper system.
I did manage to fix the original problem though of instability with decimation of 4. Turns out I had substeps set to 2 instead of 1 in the simulator parameters.