Improbable-AI/walk-these-ways #44: When recording data using L2 button, generated data does not contain "tau_est"

Robokan, 2023-07-14:
I trained and deployed my own policy. Then used L2 button to start and then stop recording data. Then tried to train an actuator net with the data but it doesn't contain any "tau_est" values so it fails on line 166:

 if "tau_est" not in datas[0].keys():
        return

How do I get it to record the "tau_est"?

gmargo11, 2023-07-18:
Hi @Robokan ,

The entries in the `infos` dictionary here will be logged: https://github.com/Improbable-AI/walk-these-ways/blob/master/go1_gym_deploy/envs/lcm_agent.py#L282

The latest value of `tau_est` is maintained as an attribute of the `StateEstimator`: https://github.com/Improbable-AI/walk-these-ways/blob/master/go1_gym_deploy/utils/cheetah_state_estimator.py#L270

So, adding the following after line 297 of `lcm_agent.py` should work: 

```
"tau_est": self.se.tau_est,
```

-Gabe

closed 2023-07-18 by gmargo11

Robokan, 2023-07-20:
Yep, that works. Thanks.

Robokan, 2023-07-21:
How much data do a I need to record? I ran my robot though all of its movements and then trained the network yet it was unable to match it's the torques.

Robokan, 2023-07-23:
I think I figured it out. I needed to record for longer and I also have a strange vibration that only appears when I have transferred the controller to the robot. I think if I can figure out the vibration this will train better.