Improbable-AI/walk-these-ways #45: Code not running on Untree A1 robot

Shifters1, 2023-07-23:
Thank you for your work. I have been trying to deploy your policy on a unitree A1 robot but it doesnt work. I have followed all your instruction as specified on your page. I editted the Lcm_position.cpp to suit UDP requirement  of A1 and i have compiled and updated the lcm_position file. I have verified the deployemt.tar file to ensure that the check md5sum deployment_image.tar  has 
ca70e4a10063fa39f026d32b1fe5d88b. While troubleshooting this problem, I came across a similar issue on your webpage https://github.com/Improbable-AI/walk-these-ways/issues/1] and it still doesnt provide any solution.

Each time  try to run this code, i keep getting stuck at the image below
[image]

[image]

Please help with a solition to this!!

gmargo11, 2023-08-02:
Hi @Shifters1 ,

Unfortunately, I don't have access to an A1 robot, so I can only give some high level suggestions.

After running the `lcm_position` file, try moving the joints. Do they move freely or resist motion? If they move freely, there is a problem with the `lcm_position` file and its usage of the `unitree_legged_sdk`. If they resist motion, there is a problem with the communication between `lcm_position` and the policy code.

If the former, make sure you can run the low-level examples from the `unitree_legged_sdk` repo.

If the latter, try entering the docker container in interactive mode using `sudo make run` instead of `sudo make autostart`. Then run the deployment script inside docker in interactive mode: `cd /home/isaac/go1_gym/ && python3 setup.py develop && cd go1_gym_deploy/scripts && ls && python3 deploy_policy.py`. Do you see any additional output?

Shifters1, 2023-08-03:
Thank you for your prompt response. I successfully resolved the issue by commenting out the code section that required me pressing R2, enabling the policy to function on the Unitree A1 robot. However, a new problem arose as the remote control is now non-responsive in low level, although it works generally. Additionally, i tested the default example_joystick on the SDK and it showed no response when pressing A. As a result, I currently lack the capability to control the gaiting and fully utilise your work. 

Furthermore, I noticed that terminating the policy any time is running on the robot is not feasible, it always requires shutting down the robot.

Could you help with possible solutions?

Many thanks Gabe

gmargo11, 2023-08-03:
Hmm, if` example_joystick` is not working, this is a problem with Unitree's SDK on A1. Unfortunately, I can only suggest opening an issue in https://github.com/unitreerobotics/unitree_legged_sdk or emailing Unitree support.

By the way, are you using v3.3.1 of the unitree_legged_sdk? I noticed in https://github.com/unitreerobotics/unitree_legged_sdk#readme that this version is required for A1.

-Gabe

Shifters1, 2023-08-04:
Thank you  Gabe for your quick response and suggested solutions. We are currently using v3.3.1 and have discovered that the problem is related to the SDK. We have already contacted the Unitree support service.

However, there is one more question that you omitted. We're having trouble stopping the policy once it's starts running on the robot which always necessitate total shutdown of the robot. Is there a known solution to this issue?

gmargo11, 2023-08-05:
If you are running docker in interactive mode as described above, you can kill the controller with `Ctrl-C` in the docker window. 

If you just want to stop the robot, the button is [R2]. But since your controller's not working, this won't work for you yet.

By the way, because this is a common request, I will be pushing a new `pretrain-v1` policy soon that automatically transitions to a learned standing mode when the velocity command is low. This will mean the robot doesn't trot in place anymore.

-Gabe

keving2733, 2023-08-15:
Hello @Shifters1, I am also trying to deploy a trained policy on an A1. What specifically needs to change in Lcm_position.cpp to suit UDP requirement of A1? Is there anything else that needs to be changed in this walk_these_ways package? I've looked at the unitree_legged_sdk for the A1 and not sure if that is supposed to help, but if you could point me to any other resources I would greatly appreciate it, thanks!

Shifters1, 2023-08-16:
Hi @keving2733 

Here is how we deployed ours. I downloaded  and built SDK Version 3.3.1 on the robot and then updated the sport version is 1.19 on the Upboard(board toward the hind leg). Copy the lcm_position.cpp from walk these ways  to the example file of the SDK, comment out . the reason is because battery management/ bms wasnt capture in V 3.3.1.  `battery_state_simple.current = state.bms.current;
    battery_state_simple.cycle = state.bms.cycle;
    battery_state_simple.SOC = state.bms.SOC;
    battery_state_simple.bms_status = state.bms.bms_status;
    for(int i = 0; i < 10; i++){
        battery_state_simple.cell_voltage[i] = state.bms.cell_vol[i];
    }

    _simpleLCM.publish("battery_state", &battery_state_simple);`

closed 2023-08-16 by Shifters1

keving2733, 2023-08-29:
Hello @Shifters1,

You copied the lcm_position.cpp into the "example" folder correct? I am also unable to identify where in "lcm_position.cpp" the parts about the battery are. I also notice in the sdk in the "example" directory the example files seem to name the robot "Go1". I may be looking in the wrong place but if you happen to know where it is where the robot is named, I would greatly appreciate it. Thanks!

Best,

Kevin Garcia

Shifters1, 2024-02-13:
Hi Gabe , 

I would like to know when you plan to release the pretrain-v1 policy. Thank you

> 3. By the way, because this is a common request, I will be pushing a new `pretrain-v1` policy soon that automatically transitions to a learned standing mode when the velocity command is low. This will mean the robot doesn't trot in place anymore.

D-jojo, 2024-09-10:
In the deploy_policy interface, 'not start' appears at the bottom. Is this related to the robot not moving
[image]
[image]