Improbable-AI/walk-these-ways #47: open /var/lib/docker/tmp/docker-import-221834566/repositories: no such file or directory

SDU-KIKI, 2023-08-03:
When I was trying to install the docker to unitree-desktop by the command: sudo ../installer/install_deployment_code.sh, the error as the title shown has happened. Have some convenient ways to solve  this problem? Thanks a lot!

gmargo11, 2023-08-03:
Hi @SDU-KIKI ,

Another user encountered the same error before: https://github.com/Improbable-AI/walk-these-ways/issues/42

Check the md5sum of `deployment_image.tar` and follow the other suggestions in that issue, then let me know if the problem persists?

-Gabe

closed 2023-08-03 by gmargo11

SDU-KIKI, 2023-08-04:
Thank you for your solution！I have settled this error down by deleting the deployment_image.tar I download before, which was 0 byte due to the network problem. Then I retried to download it and was successful.