Improbable-AI/walk-these-ways #48: ./lcm_position: error while loading shared libraries: liblcm.so.1: cannot open shared object file: No such file or directory

SDU-KIKI, 2023-08-04:
When I tried to start the controller with the command: ./start_unitree_sdk.sh, the problem happened. It seemed that the docker haven't installed the lcm correctly?

gmargo11, 2023-08-05:
Hi @SDU-KIKI ,

It sounds like you may be running `./start_unitree_sdk.sh` inside docker. If so, please try running it outside of docker and let me know if the problem persists. The script is intended to be run on the Jetson Nano's host operating system.

-Gabe

SDU-KIKI, 2023-08-07:
I tried running it inside docker by :
python3 setup.py install
cd go1_gym_deploy/scripts
python3 deploy_policy.py
When it print the message: "About to calibrate; the robot will stand [Press R2 to calibrate]", I pressed the [R2] of RC, but nothing happened after a while.

amrmousa144, 2023-08-17:
Add this line to your `.bashrc` to append the path of your venv's lib folder to the system:
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/your/venv/lib`

closed 2023-10-25 by gmargo11

Anchoret13, 2024-01-17:
> I tried running it inside docker by : python3 setup.py install cd go1_gym_deploy/scripts python3 deploy_policy.py When it print the message: "About to calibrate; the robot will stand [Press R2 to calibrate]", I pressed the [R2] of RC, but nothing happened after a while.

Hi @SDU-KIKI , I met the same problem. How did you solve that?