Improbable-AI/walk-these-ways #50: When we change some terrain in this situation why robots are start at sky?

desktop-ls, 2023-09-19:
Hi. 

I have some problem.

when we use simulation in legged_gym(original learning in minuites code), on downstairs or upstairs, robots initial position is Almost stuck. i mean like below 
[image]

But in this walk-these-ways code express like below
[image]

Which part should i change? 
I really want to know it thank you.

OlalekanIsola, 2024-02-06:
Could you please point me to where in the code you added these terrains. I am trying to add more terrains  to the flat  which came by default. Many thanks

OlalekanIsola, 2024-02-07:
I solved it by modifying 'terrain_proportions' in the 'go1_config.py' file. Note that the sum of the values in the list should be equal to 1. Feel free to adjust each terrain to determine how much you want it to appear.

Regarding the floating robot issue, you may want to check your 'pos'. Try using the following for 'go1': [0.0, 0.0, 0.34]. If you have a disjointed urdf, try changing "flip_visual_attachments" to either True or False