Improbable-AI/walk-these-ways #52: keeping it from shutting down when it falls over to allow a get up behavior to work

Robokan, 2023-09-30:
I have created a NN with a getup behavior but when it falls it just shuts down. How do I keep it from shutting off when it falls over?

gmargo11, 2023-10-25:
Hi @Robokan,

The orientation emergency stop is triggered by lines 159-162 in `deployment_runner.py`:

```
# bad orientation emergency stop
rpy = self.agents[self.control_agent_name].se.get_rpy()
if abs(rpy[0]) > 1.6 or abs(rpy[1]) > 1.6:
    self.calibrate(wait=False, low=True)
```

You can turn it off by commenting out those lines here: https://github.com/Improbable-AI/walk-these-ways/blob/master/go1_gym_deploy/utils/deployment_runner.py#L159

-Gabe

closed 2023-10-25 by gmargo11

Robokan, 2023-10-29:
Thanks!

This is my deep learning motion capture project up and walking on the Go1 thanks to your help:
see:
https://www.youtube.com/playlist?list=PLRomMJJB2S8ZIspr5rFre0uzoUa4taO42

gmargo11, 2023-10-29:
@Robokan, these videos are very impressive! It's clear that you built some significant new tools to get this working. I'm glad if our code and my replies on here helped a bit :)

Did you end up using a custom actuator network -- I remember you were collecting some new data for this? And are you using a technique like AMP to specify the imitation task?