Improbable-AI/walk-these-ways #59: Did this require that I buy one unitree robot dog first?

tula-liuxin, 2023-12-29:
I saw it mentioned I should connect unitree serve, maybe not? Should I get one unitree dog as requirement?

gmargo11, 2023-12-29:
Hi @tula-liuxin ,

Glad you're enjoying the repo!

**You can run the controller in simulation first; no dog is required.**

You'll be able to train and play the walk-these-ways controller for a simulated robot by following the instructions in "Training a Model" section: https://github.com/Improbable-AI/walk-these-ways?tab=readme-ov-file#training-a-model- . You don't need any physical robot for this stage. The next stage, "Deploying a Model", is the only stage that requires a physical Unitree Go1 Edu robot. 

-Gabe

closed 2023-12-29 by gmargo11

tula-liuxin, 2023-12-30:
Thank you for your reponse! It's helpful