Improbable-AI/walk-these-ways #64: Deploy on go2 edu

Simonforyou, 2024-01-10:
I want to deploy this algorithm on go2 edu, but I found that the sdk version has been updated. Is lcm not supported in the new version of sdk?

gmargo11, 2024-01-30:
Hi @Simonforyou 

Yes, you'll need to modify and rebuild the `lcm_position.cpp` file to use the new SDK version. I think the process should be similar to adapting the code to A1: https://github.com/Improbable-AI/walk-these-ways/issues/45#issuecomment-1680176221

Unfortunately I don't have a Go2 yet so I can't add support myself. If you or any other users succeed, definitely submit a pull request!

Best
Gabe

zeonsunlightyu, 2024-04-09:
Hi,guys,here is my solution for go2edu sim2real ,hope it can helps people who still struggling 

https://github.com/zeonsunlightyu/LocomotionWithNP3O/tree/master

I will post a video link to readme later on 

(the quality of code is pretty bad at the moment, but it do works, it will improve over time)

gmargo11, 2024-06-16:
Cool, nice work @zeonsunlightyu and thanks for sharing it here!

I also noticed this repository porting walk-these-ways to Go2, which may be relevant to folks browsing this issue: https://github.com/Teddy-Liao/walk-these-ways-go2 (the work is done independently by others, I'm not affiliated with this project)

closed 2024-06-16 by gmargo11