Improbable-AI/walk-these-ways #71: error 'sudo make autostart' command in rasberrypi4(arm32)

dongjukim123, 2024-03-06:
Hi I am goint to do project implementing the code to a1. The board is rasberry pi4(arm32bit) I succeeded
 `cd ~/go1_gym/go1_gym_deploy/autostart  ./start_unitree_sdk.sh` changing cmakelist with 32bit arm .But there was some issue in the
`cd ~/go1_gym/go1_gym_deploy/docker
sudo make autostart`
with this error
[image]
I think this error occurred because of OS(I used rasberry pi4)
I tried to delete --runtime =nvidia and --net =host But other issues was coming... 
Do you know how to solve this problem and fix error?..
In my thought, I have to change all dockerfile to fit arm32(rasberrypi 4)

gmargo11, 2024-06-16:
Hi @dongjukim123 ,

I haven't built the image on Pi before. I think you're correct that you'll need to change the dockerfile base image to match the Pi's architecture and also remove the `--runtime=nvidia` flag since the Pi does not have an NVIDIA GPU. I think this should resolve the invalid runtime error. If you post the subsequent errors you mentioned, I can try to advise.

-Gabe

closed 2025-02-17 by gmargo11

sanbens1234, 2025-03-15:
> Hi I am goint to do project implementing the code to a1. The board is rasberry pi4(arm32bit) I succeeded `cd ~/go1_gym/go1_gym_deploy/autostart ./start_unitree_sdk.sh` changing cmakelist with 32bit [...]

hi, I wonder if you solve this problem? I face the same issues.