Improbable-AI/walk-these-ways #75: Problem with recompiling LCM_position

hanzhi0410, 2024-03-23:
When I used A1's SDK to regenerate LCM_position, the following issues occurred

/home/hn/legged_RL/walk-these-ways/a1_gym_deploy/unitree_legged_sdk_bin/lcm_position.cpp: In constructor ‘Custom::Custom(uint8_t)’:
/home/hn/legged_RL/walk-these-ways/a1_gym_deploy/unitree_legged_sdk_bin/lcm_position.cpp:24:67: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
   24 |     Custom(uint8_t level): safe(LeggedType::A1), udp(level, 8090, "192.168.123.10", 8007) {
      |                                                                   ^~~~~~~~~~~~~~~~
      |                                                                   |
      |                                                                   const char*
In file included from /home/hn/legged_RL/walk-these-ways/a1_gym_deploy/unitree_legged_sdk_bin/unitree_legged_sdk/unitree_legged_sdk.h:11,
                 from /home/hn/legged_RL/walk-these-ways/a1_gym_deploy/unitree_legged_sdk_bin/lcm_position.cpp:5:
/home/hn/legged_RL/walk-these-ways/a1_gym_deploy/unitree_legged_sdk_bin/unitree_legged_sdk/udp.h:26:54: note:   initializing argument 3 of ‘UNITREE_LEGGED_SDK::UDP::UDP(uint16_t, uint16_t, int, int)’
   26 |     UDP(uint16_t localPort, uint16_t targetPort, int sendLength, int recvLength); // as server, client IP can change
      |                                                  ~~~~^~~~~~~~~~
make[2]: *** [CMakeFiles/lcm_position.dir/build.make:63：CMakeFiles/lcm_position.dir/unitree_legged_sdk_bin/lcm_position.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:105：CMakeFiles/lcm_position.dir/all] 错误 2
make: *** [Makefile:84：all] 错误 2

I have reviewed the relevant udp. h file again，discovered corresponding content in sdk-3.8.0 used by go1.
namespace UNITREE_LEGGED_SDK
{
    constexpr int UDP_CLIENT_PORT = 8080;                       // local port
    constexpr int UDP_SERVER_PORT = 8007;                       // target port
    constexpr char UDP_SERVER_IP_BASIC[] = "192.168.123.10";    // target IP address
    constexpr char UDP_SERVER_IP_SPORT[] = "192.168.123.161";   // target IP address

    typedef enum {
        nonBlock     = 0x00,
        block        = 0x01,
        blockTimeout = 0x02,
    } RecvEnum;

    // Notice: User defined data(like struct) should add crc(4Byte) at the end.
    class UDP {
	public:
        UDP(uint8_t level, uint16_t localPort, const char* targetIP, uint16_t targetPort);  // udp use dafault length according to level
        UDP(uint16_t localPort, const char* targetIP, uint16_t targetPort, 
            int sendLength, int recvLength, bool initiativeDisconnect = false, RecvEnum recvType = RecvEnum::nonBlock);
        UDP(uint16_t localPort, 
            int sendLength, int recvLength, bool initiativeDisconnect = false, RecvEnum recvType = RecvEnum::nonBlock, bool setIpPort = false);  
        ~UDP();

But the content in my SDK-3.2 has changed as follows
namespace UNITREE_LEGGED_SDK
{
constexpr int UDP_CLIENT_PORT = 8090;                       // local port
constexpr int UDP_SERVER_PORT = 8007;                       // target port
constexpr char UDP_SERVER_IP_BASIC[] = "192.168.123.10";    // target IP address
constexpr char UDP_SERVER_IP_SPORT[] = "192.168.123.161";   // target IP address

// Notice: User defined data(like struct) should add crc(4Byte) at the end.
class UDP {
public:
    UDP(uint8_t level, HighLevelType highControl = HighLevelType::Basic);  // unitree dafault IP and Port
    UDP(uint16_t localPort, const char* targetIP, uint16_t targetPort, int sendLength, int recvLength);
    UDP(uint16_t localPort, uint16_t targetPort, int sendLength, int recvLength); // as server, client IP can change
    ~UDP();

Do you have a good solution

hanzhi0410, 2024-03-23:
I have found a solution by changing LCM to Custom (uint8_t level): safe (LeggedType:: A1), udp (level)

gmargo11, 2024-06-16:
Glad you found a solution, @hanzhi0410 ! Hope you found the repository useful in your work on A1.

-Gabe

closed 2024-06-16 by gmargo11