Improbable-AI/walk-these-ways #79: something wrong when "sudo make autostart"

PoirotJin, 2024-06-04:
Because of some reasons, I have to use python3.8. When I run the command "sudo make autostart", I get the fault: 
```
Traceback (most recent call last):
  File "deploy_policy.py", line 6, in <module>
    from go1_gym_deploy.utils.deployment_runner import DeploymentRunner
  File "/usr/local/lib/python3.6/dist-packages/go1_gym-1.0.0-py3.6.egg/go1_gym_deploy/utils/deployment_runner.py", line 6, in <module>
    import torch
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 196, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 149, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [libcurand.so](http://libcurand.so/).10: cannot open shared object file: No such file or directory
make: *** [Makefile:27: autostart] Error 1
unitree@ubuntu:~/go1_gym/go1_gym_deploy/docker$ ls /usr/local/lib/python3.6/dist-packages/
ls: cannot access '/usr/local/lib/python3.6/dist-packages/': No such file or directory
```
So how can I run it on python3.8?

gmargo11, 2025-02-17:
Hi @PoirotJin ,

`sudo make autostart` run code inside a docker container. This references a separate filesystem from your host machine's hence why `ls /usr/local/lib/python3.6/dist-packages/` turns up empty. The issue seems related to accessing cuda libraries from the docker container. Are you running on a machine with CUDA?

-Gabe

PoirotJin, 2025-02-18:
I got it. The code can run as you describe. Thanks!

closed 2025-02-18 by PoirotJin