Improbable-AI/walk-these-ways #85: How to generate .jit files in other projects (Go2) that can be used for that project's deployment

make121, 2024-10-29:
I have implemented deployment on go2 and would like to be able to conveniently deploy training models based on other algorithms to go2. But the jit file generation process is confusing, and I don't know how to generate .jit files in another project that can be deployed here. Looking forward to your reply.

make121, 2024-11-01:
And I would also like to know how the other files under runs are generated to be imported properly in the project.

myoooofall, 2024-12-11:
> I have implemented deployment on go2 and would like to be able to conveniently deploy training models based on other algorithms to go2. But the jit file generation process is confusing, and I don't [...]

hey buddy ,i am facing the same problem with you,so can i know how you solve the problem.

gmargo11, 2025-02-17:
Hi @make121 @myoooofall , another project has recently ported our code to Go2, please find the relevant resources here: https://github.com/Teddy-Liao/walk-these-ways-go2

-Gabe

closed 2025-02-17 by gmargo11