Improbable-AI/walk-these-ways #86: how to deploy walk-these-ways to unitree B1 robot?

zdpisadaniu, 2024-12-04:
Dear Sir:
       I am so grateful to you to make such a  great job ! I deploy this project to unitree go2 robot successfully. However ,if i train a more bigger robot like Unitree B1 robot .Could you offer some suggestions about tuning the corresponding variables ?Or say ,i should pay attention to which parameters?

Best regards!

NirajPudasaini, 2025-01-24:
@zdpisadaniu any solution?

gmargo11, 2025-02-17:
Hi @zdpisadaniu , 

We actually have another project repository that applies a similar training process to the B1 robot: https://github.com/Improbable-AI/learning-compliance . It has some special additional features for controlling the arm, but the locomotion-related parameters are retuned for B1 and can be a good reference. The parameters I would focus on adjusting after loading the new robot model are:
- PD gains (increase for heavier robot)
- Reward weights (esp. for terms containing torque, since the torques of B1 will be higher)
- Max torque and velocity limits (similar reason to above)
- Robot-specific reward parameters like the base height range, footswing height range, stance width and length. Also, the best gait frequency may generally be lower for B1

All those parameters should have reference ranges in the linked learning-compliance repo.

Best luck with the project!

-Gabe

closed 2025-02-17 by gmargo11

zdpisadaniu, 2025-02-18:
Thank you very much!!!